tonyp7/esp32-nixie-clock #1: Missing source files

Blaster1920, 2020-03-01:
Hi there,
I was interested in replicating your project for personal use, but I couldn't find anywhere in the repo the source files for the PCB and schematics.
Would you be able to upload them?
Thanks in advance.

tonyp7, 2020-03-02:
Hello,

Yes I need to upload a few more files --including pictures of the final clock.  I'll do that this week!

Cheers

tonyp7, 2020-03-02:
I added gerber files. I don't understand your comment about schematics though, it's all there (?)

Blaster1920, 2020-03-02:
Hi there,
thanks for the reply and the gerbers.
What I meant is that it's missing the source files for the schematics,  the proprietary files of the CAD you've used (eg. Kicad .sch).

tonyp7, 2020-03-07:
I pushed the original file (Dip Trace) in /schematics and added the BOM. Good luck!

Blaster1920, 2020-03-07:
> I pushed the original file (Dip Trace) in /schematics and added the BOM. Good luck!

Thanks for your help! I'll give it a try.

closed 2020-07-21 by tonyp7