tonyp7/esp32-nixie-clock #2: Is this project open source?

rudinikun, 2020-07-28:
Hi Tony,

I see you are working on an extraordinary project. I want to ask if this project is open source, in the sense that I can print the PCB that you made?

Thank you,
Rudini

tonyp7, 2020-07-28:
Absolutely yes it is open source! You can have the PCB made freely.

Also I know this repository is a bit bare right now but that's because it's mostly a personal project. I'll add some pictures soon!

rudinikun, 2020-07-28:
Awesome, can't wait for that. Also can you please share the BOM for clock face. Thank you Tony

tonyp7, 2020-07-29:
BOM is added

tonyp7, 2020-07-29:
I realize it's missing the WS2812 leds which were added later -- I need to update the schematics and BOM to match the PCB

rudinikun, 2020-07-30:
Hi Tony, 

Thank you for adding BOM. I am very unfamiliar with hardware, if I print a PCB and install all electronic components, the next step is to upload the code into ESP32, is that right Tony, Are there other devices that must be prepared, please advise?

Thank you,
Rudini

tonyp7, 2020-07-30:
That's right. It's pretty much assemble the PCBs and upload the firmware to ESP32. 

That being said if you are a beginner in that space I'd strongly advise you to not start with this. This project is still very much work in progress so the code is not complete; and the PCBs themselves might be challenging as they contain very small parts (0402 and 0.5mm pitch components).

closed 2020-08-01 by tonyp7